Pioreactor/pioreactor #63: avahi-alias service fails

CamDavidsonPilon, 2021-02-08:
On reboot: 

```
● avahi-alias@pioreactor.local.service - Publish pioreactor.local as alias for pioreactor1.local via mdns
   Loaded: loaded (/lib/systemd/system/avahi-alias@.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2021-02-08 20:37:03 GMT; 1h 4min ago
  Process: 354 ExecStart=/bin/bash -c /usr/bin/avahi-publish -a -R pioreactor.local $(avahi-resolve -4 -n pioreactor1.local | cut -f 2) (code=exited, status=1
 Main PID: 354 (code=exited, status=1/FAILURE)

Feb 08 20:36:57 pioreactor1 systemd[1]: Started Publish pioreactor.local as alias for pioreactor1.local via mdns.
Feb 08 20:37:03 pioreactor1 bash[354]: Failed to resolve host name 'pioreactor1.local': Timeout reached
Feb 08 20:37:03 pioreactor1 bash[354]: Bad number of arguments
Feb 08 20:37:03 pioreactor1 systemd[1]: avahi-alias@pioreactor.local.service: Main process exited, code=exited, status=1/FAILURE
Feb 08 20:37:03 pioreactor1 systemd[1]: avahi-alias@pioreactor.local.service: Failed with result 'exit-code'.
```

CamDavidsonPilon, 2021-02-12:
I added a retry block, will succeed eventually

closed 2021-02-12 by CamDavidsonPilon

HiroTamori, 2021-11-09:
I put an example for someone who has the same trouble:

```
[Unit]
Description=Publish %I as alias for %H.local via mdns

[Service]
Type=simple
ExecStart=/bin/bash -c "/usr/bin/avahi-publish -a -R %I $(hostname -I | cut -f1 -d' ')"
Restart=on-failure
RestartSec=5s

[Install]
WantedBy=multi-user.target
```

and enable service with instance name like;
```
sudo systemctl enable --now avahi-alias@myhost.local
```

CamDavidsonPilon, 2021-11-09:
Nice suggestion @HiroTamori!