Pioreactor/pioreactor #238: Failed to star FirstBoot

VitorFrost, 2022-01-21:
If it keeps failing, try running
```
journalctl -u firstboot.service
```

and it may give more details about why it failed

_Originally posted by @CamDavidsonPilon in https://github.com/Pioreactor/pioreactorui/issues/86#issuecomment-1018660503_

```
Journal file /var/log/journal/34681e7479df401b9a368f438806e522/system@0005cf91300d9a89-ae7f92bdccdbc20d.journal~ is truncated, ignoring file.
-- Journal begins at Sat 2021-10-30 08:29:41 -03, ends at Sat 2021-10-30 09:56:26 -03. --
Oct 30 08:30:03 pioreactor systemd[1]: Starting FirstBoot...
Oct 30 08:30:03 pioreactor firstboot.sh[446]: + set -e
Oct 30 08:30:03 pioreactor firstboot.sh[446]: + export LC_ALL=C
Oct 30 08:30:03 pioreactor firstboot.sh[446]: + LC_ALL=C
Oct 30 08:30:03 pioreactor firstboot.sh[446]: + PIO_DIR=/home/pi/.pioreactor
Oct 30 08:30:03 pioreactor firstboot.sh[446]: + sudo -u pi ssh-keygen -q -t rsa -N '' -f /home/pi/.ssh/id_rsa
Oct 30 08:30:06 pioreactor sudo[449]:     root : PWD=/ ; USER=pi ; COMMAND=/usr/bin/ssh-keygen -q -t rsa -N  -f /home/p>
Oct 30 08:30:06 pioreactor sudo[449]: pam_unix(sudo:session): session opened for user pi(uid=1000) by (uid=0)
Oct 30 08:30:18 pioreactor firstboot.sh[446]: + sudo -u pi cat /home/pi/.ssh/id_rsa.pub
Oct 30 08:30:18 pioreactor firstboot.sh[764]: ++ hostname
Oct 30 08:30:18 pioreactor firstboot.sh[446]: + sudo -u pi ssh-keyscan -H pioreactor
Oct 30 08:30:18 pioreactor sudo[765]:     root : PWD=/ ; USER=pi ; COMMAND=/usr/bin/ssh-keyscan -H pioreactor
Oct 30 08:30:18 pioreactor sudo[765]: pam_unix(sudo:session): session opened for user pi(uid=1000) by (uid=0)
Oct 30 08:30:19 pioreactor firstboot.sh[766]: write (pioreactor): Connection refused
Oct 30 08:30:19 pioreactor firstboot.sh[766]: write (pioreactor): Connection refused
Oct 30 08:30:19 pioreactor firstboot.sh[766]: write (pioreactor): Connection refused
Oct 30 08:30:19 pioreactor firstboot.sh[766]: write (pioreactor): Connection refused
Oct 30 08:30:19 pioreactor firstboot.sh[766]: write (pioreactor): Connection refused
Oct 30 08:30:19 pioreactor sudo[765]: pam_unix(sudo:session): session closed for user pi
Oct 30 08:30:20 pioreactor systemd[1]: firstboot.service: Main process exited, code=exited, status=1/FAILURE
Oct 30 08:30:20 pioreactor systemd[1]: firstboot.service: Failed with result 'exit-code'.
Oct 30 08:30:20 pioreactor systemd[1]: Failed to start FirstBoot.
Oct 30 08:30:20 pioreactor systemd[1]: firstboot.service: Consumed 11.494s CPU time.
-- Boot 536fc477a5414b0abb34557f68dc0968 --
Oct 30 09:17:06 pioreactor systemd[1]: Starting FirstBoot...
Oct 30 09:17:06 pioreactor firstboot.sh[423]: + set -e
Oct 30 09:17:06 pioreactor firstboot.sh[423]: + export LC_ALL=C
Oct 30 09:17:06 pioreactor firstboot.sh[423]: + LC_ALL=C
Oct 30 09:17:06 pioreactor firstboot.sh[423]: + PIO_DIR=/home/pi/.pioreactor
Oct 30 09:17:06 pioreactor firstboot.sh[423]: + sudo -u pi ssh-keygen -q -t rsa -N '' -f /home/pi/.ssh/id_rsa
Oct 30 09:17:06 pioreactor sudo[427]:     root : PWD=/ ; USER=pi ; COMMAND=/usr/bin/ssh-keygen -q -t rsa -N  -f /home/p>
Oct 30 09:17:06 pioreactor sudo[427]: pam_unix(sudo:session): session opened for user pi(uid=1000) by (uid=0)
Oct 30 09:17:12 pioreactor firstboot.sh[440]: /home/pi/.ssh/id_rsa already exists.
Oct 30 09:17:12 pioreactor firstboot.sh[440]: Overwrite (y/n)?
Oct 30 09:17:12 pioreactor systemd[1]: firstboot.service: Main process exited, code=exited, status=1/FAILURE
Oct 30 09:17:12 pioreactor systemd[1]: firstboot.service: Failed with result 'exit-code'.
Oct 30 09:17:12 pioreactor systemd[1]: Failed to start FirstBoot.
Oct 30 09:17:12 pioreactor systemd[1]: firstboot.service: Consumed 5.929s CPU time.
-- Boot a0330c8df3274bdd8404755ec35f4a17 --
Oct 30 09:36:11 pioreactor systemd[1]: Starting FirstBoot...
Oct 30 09:36:11 pioreactor firstboot.sh[416]: + set -e
Oct 30 09:36:11 pioreactor firstboot.sh[416]: + export LC_ALL=C
Oct 30 09:36:11 pioreactor firstboot.sh[416]: + LC_ALL=C
Oct 30 09:36:11 pioreactor firstboot.sh[416]: + PIO_DIR=/home/pi/.pioreactor
Oct 30 09:36:11 pioreactor firstboot.sh[416]: + sudo -u pi ssh-keygen -q -t rsa -N '' -f /home/pi/.ssh/id_rsa
Oct 30 09:36:11 pioreactor sudo[421]:     root : PWD=/ ; USER=pi ; COMMAND=/usr/bin/ssh-keygen -q -t rsa -N  -f /home/p>
Oct 30 09:36:11 pioreactor sudo[421]: pam_unix(sudo:session): session opened for user pi(uid=1000) by (uid=0)
Oct 30 09:36:32 pioreactor firstboot.sh[434]: /home/pi/.ssh/id_rsa already exists.
Oct 30 09:36:32 pioreactor sudo[421]: pam_unix(sudo:session): session closed for user pi
Oct 30 09:36:32 pioreactor firstboot.sh[434]: Overwrite (y/n)?
Oct 30 09:36:32 pioreactor systemd[1]: firstboot.service: Main process exited, code=exited, status=1/FAILURE
Oct 30 09:36:32 pioreactor systemd[1]: firstboot.service: Failed with result 'exit-code'.
Oct 30 09:36:32 pioreactor systemd[1]: Failed to start FirstBoot.
Oct 30 09:36:32 pioreactor systemd[1]: firstboot.service: Consumed 21.010s CPU time.
-- Boot 810c6a2f3e5c46bb82614297ce6ef768 --
Oct 30 09:49:59 pioreactor systemd[1]: Starting FirstBoot...
Oct 30 09:49:59 pioreactor firstboot.sh[423]: + set -e
Oct 30 09:49:59 pioreactor firstboot.sh[423]: + export LC_ALL=C
Oct 30 09:49:59 pioreactor firstboot.sh[423]: + LC_ALL=C
Oct 30 09:49:59 pioreactor firstboot.sh[423]: + PIO_DIR=/home/pi/.pioreactor
Oct 30 09:49:59 pioreactor firstboot.sh[423]: + sudo -u pi ssh-keygen -q -t rsa -N '' -f /home/pi/.ssh/id_rsa
Oct 30 09:49:59 pioreactor sudo[427]:     root : PWD=/ ; USER=pi ; COMMAND=/usr/bin/ssh-keygen -q -t rsa -N  -f /home/p>
Oct 30 09:49:59 pioreactor sudo[427]: pam_unix(sudo:session): session opened for user pi(uid=1000) by (uid=0)
Oct 30 09:50:03 pioreactor firstboot.sh[443]: /home/pi/.ssh/id_rsa already exists.
Oct 30 09:50:03 pioreactor firstboot.sh[443]: Overwrite (y/n)?
Oct 30 09:50:03 pioreactor systemd[1]: firstboot.service: Main process exited, code=exited, status=1/FAILURE
Oct 30 09:50:03 pioreactor sudo[427]: pam_unix(sudo:session): session closed for user pi
Oct 30 09:50:03 pioreactor systemd[1]: firstboot.service: Failed with result 'exit-code'.
Oct 30 09:50:03 pioreactor systemd[1]: Failed to start FirstBoot.
Oct 30 09:50:03 pioreactor systemd[1]: firstboot.service: Consumed 4.107s CPU time.
```

CamDavidsonPilon, 2022-01-21:
Great, this really helps. 

While I get that sorted, you can run the rest of the commands [here](https://github.com/Pioreactor/CustoPiZer/blob/pioreactor/workspace/scripts/files/bash/firstboot_leader.sh#L16-L34) from the RPi's command line:

```
crudini --set $PIO_DIR/config.ini network.topology leader_hostname $(hostname)
crudini --set $PIO_DIR/config.ini network.topology leader_address $(hostname).local

mosquitto_pub -t "pioreactor/latest_experiment" -m "Demo experiment" -r

sudo systemctl restart avahi-alias.service

echo "0 0 */5 * * /usr/local/bin/pio run backup_database" | crontab -
```

I'll try to find a proper solution in the meantime

CamDavidsonPilon, 2022-01-21:
Some questions when you get a chance:

1. Are you on a university, work, or home network?
2. Does `ssh-keyscan pioreactor` give an error?
3. Does `ssh-keyscan -4 pioreactor` give an error?
4. Does `ssh pioreactor` give an error? (It may prompt you, select 'yes')

The UI may or may not be working 100%. If it's _not_ working, try the following line:

```
sudo -u pi echo "StrictHostKeyChecking no" >> ~/.ssh/config
```

and try the UI again.

VitorFrost, 2022-01-22:
> Some questions when you get a chance:
> 
>     1. Are you on a university, work, or home network?
> 
>     2. Does `ssh-keyscan pioreactor` give an error?
> 
>     3. Does `ssh-keyscan -4 pioreactor` give an error?
> 
>     4. Does `ssh pioreactor` give an error? (It may prompt you, select 'yes')
> 
> 

1: home
2: no
3: no
4: yes! (kind off)
got this on 4: 
```
ssh pioreactor
The authenticity of host 'pioreactor (127.0.1.1)' can't be established.
ECDSA key fingerprint is SHA256:***************************************.
Are you sure you want to continue connecting (yes/no/[fingerprint])? 
```

Also, 
> The UI may or may not be working 100%. If it's _not_ working, try the following line:
> 
> ```
> sudo -u pi echo "StrictHostKeyChecking no" >> ~/.ssh/config
> ```
> 
> and try the UI again.

didn't fixed anything

But this did:

> Great, this really helps.
> 
> While I get that sorted, you can run the rest of the commands here from the RPi's command line:
> ```
> PIO_DIR=/home/pi/.pioreactor
> 
> 
> crudini --set $PIO_DIR/config.ini network.topology leader_hostname $(hostname)
> crudini --set $PIO_DIR/config.ini network.topology leader_address $(hostname).local
> 
> mosquitto_pub -t "pioreactor/latest_experiment" -m "Demo experiment" -r
> 
> sudo systemctl restart avahi-alias.service
> 
> echo "0 0 */5 * * /usr/local/bin/pio run backup_database" | crontab -
> ```
> I'll try to find a proper solution in the meantime

CamDavidsonPilon, 2022-04-29:
Added more logging around this. It will be much more obvious if firstboot fails to run.

closed 2022-04-29 by CamDavidsonPilon