Pioreactor/pioreactor #239: No NTP (date and time) updates (is it a localized problem?)

VitorFrost, 2022-01-22:
I could not identify if this is a general problem, or if this is a local NTP problem.
Mine doesn't get time synced. This can lead to problems, as it has a demo experiment that has started in the future, or multiple experiments won't sync.

I always have this answer from timedatectl:
```
 timedatectl
               Local time: Sat 2021-10-30 11:23:52 -03
           Universal time: Sat 2021-10-30 14:23:52 UTC
                 RTC time: n/a
                Time zone: America/Sao_Paulo (-03, -0300)
System clock synchronized: no
              NTP service: active
          RTC in local TZ: no
```

tried to force a re-sync with:
```
sudo systemctl restart systemd-timedated
sudo systemctl restart systemd-timesyncd
```

But I get a connection timeout, besides the ping response is fine:
```
 pi@pioreactor:~ $ systemctl status systemd-timesyncd
● systemd-timesyncd.service - Network Time Synchronization
     Loaded: loaded (/lib/systemd/system/systemd-timesyncd.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2021-10-30 11:31:51 -03; 1min 54s ago
       Docs: man:systemd-timesyncd.service(8)
   Main PID: 899 (systemd-timesyn)
     Status: "Idle."
      Tasks: 2 (limit: 1716)
        CPU: 229ms
     CGroup: /system.slice/systemd-timesyncd.service
             └─899 /lib/systemd/systemd-timesyncd

Oct 30 11:31:51 pioreactor systemd[1]: Starting Network Time Synchronization...
Oct 30 11:31:51 pioreactor systemd[1]: Started Network Time Synchronization.
Oct 30 11:32:01 pioreactor systemd-timesyncd[899]: Timed out waiting for reply from 200.160.7.193:123 (200.160.7.193).
Oct 30 11:32:44 pioreactor systemd-timesyncd[899]: Timed out waiting for reply from 200.160.7.193:123 (200.160.7.193).

pi@pioreactor:~ $ ping 200.160.7.193
PING 200.160.7.193 (200.160.7.193) 56(84) bytes of data.
64 bytes from 200.160.7.193: icmp_seq=1 ttl=56 time=8.37 ms
64 bytes from 200.160.7.193: icmp_seq=2 ttl=56 time=10.2 ms
64 bytes from 200.160.7.193: icmp_seq=3 ttl=56 time=15.7 ms
64 bytes from 200.160.7.193: icmp_seq=4 ttl=56 time=11.5 ms
64 bytes from 200.160.7.193: icmp_seq=5 ttl=56 time=8.79 ms
64 bytes from 200.160.7.193: icmp_seq=6 ttl=56 time=13.4 ms
64 bytes from 200.160.7.193: icmp_seq=7 ttl=56 time=8.30 ms
^C
--- 200.160.7.193 ping statistics ---
7 packets transmitted, 7 received, 0% packet loss, time 6011ms
rtt min/avg/max/mdev = 8.295/10.910/15.702/2.618 ms
```

The solution was to add a custom NTP on the file ` /etc/systemd/timesyncd.conf `
I also enabled default fallback servers, by editing the file _/etc/systemd/timesyncd.conf_
To do it, go to terminal and tipe:
´´´sudo nano  /etc/systemd/timesyncd.conf´´´

Then, edit the above lines:
```
[Time]
NTP=pool.ntp.org
FallbackNTP=0.debian.pool.ntp.org 1.debian.pool.ntp.org 2.debian.pool.ntp.org 3.debian.pool.ntp.org
```
Type Ctrl+O and Enter to save
Type Ctrl+X to Exit

Then, after a reboot, it worked for me.

CamDavidsonPilon, 2022-01-22:
> This can lead to problems, as it has a demo experiment that has started in the future

Like in your UI, it showed up as a future date? 

Yea, I need to properly code timezones better, and have documentation about them. This is a good start!

CamDavidsonPilon, 2022-01-24:
Rereading this issue, and doing more research on NTP. Okay, yes, I see the problem. I've taken for granted how in-sync the Rpi's clocks are in a cluster. Definitely something to investigate.

CamDavidsonPilon, 2022-01-24:
One more datapoint: on my local Rpi, I get:

```
               Local time: Sat 2022-01-22 22:52:16 EST
           Universal time: Sun 2022-01-23 03:52:16 UTC
                 RTC time: n/a
                Time zone: America/Toronto (EST, -0500)
System clock synchronized: yes
              NTP service: active
```

CamDavidsonPilon, 2022-01-24:
This isn't the first time your Rpi has had a network connection issue (related to the firstboot problem we saw, too). Have you seen any other network problems?

VitorFrost, 2022-01-26:
> This isn't the first time your Rpi has had a network connection issue (related to the firstboot problem we saw, too). Have you seen any other network problems?

Ok, updates on this!
I tried another setup, from the scratch. Now, from the university, with a router running exclusively for the lab.
Here I am using the pi directly connected to the router via cable.
Tested with an R.pi-2 with a fresh copy of the Pioreactor image.
On this setup, both raspberry can be found via local names (Bonjour service). This wasn't working on my home setup.
The fresh install did synced with the NPT with no problem. (I have tried it in home, with no success!) Probably something related to internet provider.

Yet, I still got the `Failed to start FirstBoot` error, and the empty page

CamDavidsonPilon, 2022-01-27:
:+1: Some improvements are coming to the image shortly that may help the firstboot problem.

CamDavidsonPilon, 2024-04-26:
We added chrony to the leaders and workers to sync time between them

closed 2024-04-26 by CamDavidsonPilon