Pioreactor/pioreactor #290: Change csv export format

kellytr, 2022-06-01:
More intuitive to create graphs from raw data 
[image]

CamDavidsonPilon, 2022-06-01:
Some thoughts: most users would want to export in a "wide" format (like displayed above), as it's much easier to work with, ideally even exported as an xsl file. However, formatting in a wide format is not trivial: 

 - needs a few packages (pandas, excel), 
 - needs to be "dataset-aware" (don't pivot for all datasets, just a listed few)

Q1. Should this be done with an app? The app would overwrite the default `pio run export_...`, and would need to install the required packages. AFAIK, no changes to the UI.

Q2. Why are ppl exporting anyways? For the ones that want to produce graphs of the data, we can perhaps build tools in the UI that make this easy: 
 - comparing multiple experiments, 
 - change the x and y range
 - adding titles,
 - etc.

CamDavidsonPilon, 2022-08-02:
This has been mostly solved by the new `pioreactor_unit_activity_data` data

closed 2022-08-02 by CamDavidsonPilon