Pioreactor/pioreactor #359: turn callback helpers in ODReader to decorators

CamDavidsonPilon, 2022-12-02:
```
from pioreactor.background_jobs.od_reading import ODReader

# example of using callbacks to OD reading...

@ODReader.add_pre_read_callback
def hello(self):
    print(f"Hello, {self}!")

@ODReader.add_post_read_callback
def goodbye(self, readings):
    print(f"Goodbye, {self}, witnessed {readings}!")

```

varuntumbe, 2022-12-03:
Hello, 

Are these issues available for public to contribute? 

Thanks

CamDavidsonPilon, 2022-12-03:
Hi @varuntumbe, 

Sure! Here are [some docs](https://docs.pioreactor.com/developer-guide/local-development) on getting your local development working. (Those docs may need some tweaks, let me know).

I work on the `develop` branch, so you can push PRs to that branch.

varuntumbe, 2022-12-04:
That's great to hear! I was looking for some challenging open source project to contribute. I will first try to install repo locally and will go forward from there. 

Thanks