Pioreactor/pioreactor #412: Implement calibration monitoring during dosing automations

odcambc, 2023-06-23:
Idea: pump/OD calibrations can be tracked for drift or errors during dosing automations. That is, (assuming vial volumes are accurate), if a given dilution fails to produce the predicted OD +/- some margin, it can throw a warning.

Might be simplest to add this to DosingAutomationJob - thoughts?

CamDavidsonPilon, 2023-06-23:
It's a good idea. Drifting pumps always worries me. Failing pumps worry me the most. A check like this could help. 

> if a given dilution fails to produce the predicted OD +/- some margin, it can throw a warning.

I feel like this is a medium-to-strong assumption, or, only works for some media types. I'm just thinking: if your media has a non-trivial turbidity to it, does this still work? (I think so but...). 

Maybe to start, the check is as simple as "did the OD decrease _at all_?". This would be an alert on a failed pump.

w.r.t. implementation: a first question to ask is "is there ever an dosing algorithm / protocol / experiment where adding from a pump expectedly _increases / non-decreases_ the OD"? I can't think of any at first thought, but 🤔

If we answer negative above, we can slip a check like this into `DosingAutomationJob` as you suggested.

odcambc, 2023-06-29:
> I feel like this is a medium-to-strong assumption, or, only works for some media types. I'm just thinking: if your media has a non-trivial turbidity to it, does this still work? (I think so but...).

Good point. Yeah, the full assumptions would be 1) the media is properly blanked, 2) vial volumes are accurate, 3) the alternative media has negligible signal. These are probably reasonable for most cases, but not universally. However, any OD targeted dosage automation wouldn't really make sense without those assumptions, and there isn't a way to have different media/alt media blanks anyway (which could be worth adding?).

Perhaps looking for an OD discontinuity is the best way to check for pump failure, and more specific tests should go in specific automations.