Pioreactor/pioreactor #540: Update to Latest Release Over Internet Leads to 500 Errors

LRParser, 2024-12-23:
I updated my v1.0 Pioreactor over the internet to the latest version (Latest version available:24.12.10) and can now no longer see any experiments, and I get 500 Internal Server Error when trying to access the routes: 
1. http://pioreactor.local/unit_api/versions/app
2. http://pioreactor.local/api/configs/config.ini
3. http://pioreactor.local/api/experiments/assignment_count
(etc)
Otherwise the GUI is partially browseable. What can I do to recover my Pioreactor? I suspect I should try to install a version from a zip, but I do not know which.
I do not particularly care about keeping the experimental data.
The shared config.ini looks like this:
```
<?xml version="1.0" encoding="iso-8859-1"?>
<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Transitional//EN"
         "http://www.w3.org/TR/xhtml1/DTD/xhtml1-transitional.dtd">
<html xmlns="http://www.w3.org/1999/xhtml" xml:lang="en" lang="en">
 <head>
  <title>500 Internal Server Error</title>
 </head>
 <body>
  <h1>500 Internal Server Error</h1>
 </body>
</html>
```

CamDavidsonPilon, 2024-12-23:
Hi @LRParser, do you know what version you were on previously? 

Report back what the following says after you SSH into your leader:

```
sudo systemctl status lighttpd.service
```

CamDavidsonPilon, 2024-12-23:
Oh, yea I know what the bug is. It's an annoying bug where your UI software version is probably up to date, but your app version is further behind. 

Honestly, depending on how far back your software version is, I would suggest just reflashing your Pioreactor with the latest image: https://docs.pioreactor.com/user-guide/software-set-up#setting-up-your-raspberry-pi

You would lose your data (but you can download the database before reflashing - let me know if you would like instructions)

Otherwise, you can solve this by going back to the previous version, and then one update at a time, getting to the latest version. Like I said, this is annoying and slow...

LRParser, 2024-12-23:
I will reflash and post here on how it goes - thank you.