Pioreactor/pioreactor #562: Opportunity to move to a new webserver

CamDavidsonPilon, 2025-05-04:
When we merge this repo and pioreactorui, and the forthcoming #460 release, we have an opportunity to change our webframework and web backend. Some ideas:

 - litestar using msgspec (!), lightweight, and is async. Actively developed, but not broad adoption. 
 - Flask (and lighttpd) is very robust, well tested, and I'm familiar with it.
 - FastAPI -> starlette + Pydantic and co., which I don't have much overlap with. This would require repeating a lot of struct definitions I think (plus might be slow with all the additional imports?)

gstrauss, 2025-05-09:
> we have an opportunity to change our webframework and web backend

What are the disadvantages with the current setup?
What are the advantages of changing?
What are the bottlenecks or problems you are trying to solve?

I am a lighttpd developer and lighttpd provides multiple ways to interfaces with backends.  Some are faster than others.  When the backend is on the local machine, then for performance and lower resource usage, I recommend using unix domain sockets to reach local backends, rather than TCP.

CamDavidsonPilon, 2025-05-09:
👋 wow cool to see a lighttp developer drop by! 

Some of the motivation is moving to more "modern" web frameworks. Flask is fine, but a lot of the Python ecosystem has moved to more strongly typed async frameworks. To use async, I would consider another web server, as lighttpd doesn't support it out of the box (I've seen the solutions though).

I'm currently in the "stay with Flask + lighttpd" camp however. I prescribe to the "choosing boring tech", and nothing is more boring than sqlite3 + lighttpd + flask!

CamDavidsonPilon, 2025-05-10:
Thank you for the information @gstrauss. Yes, asgi is what I am referring to. The move it entirely motivated by my personal DX: some modern web-frameworks in Python have really nice features, but as asgi-only. This is a case of Python web-framework-tail wagging the web-server-dog.

gstrauss, 2025-05-11:
> This is a case of Python web-framework-tail wagging the web-server-dog.

Not really.  None of this is "new".
https://asgi.readthedocs.io/en/latest/specs/main.html#version-history
ASGI version 2.0 was released 2017.
ASGI version 3.0 was released 2019.

lighttpd and nginx have been able to reverse proxy HTTP and websockets for many, many years.

As I mentioned above, I recommend using a hardened webserver such as lighttpd or nginx as the frontend to your Python doggie, whether the python backend be WSGI, ASGI, websockets, or something else.